tmiw/ezDV #19: Crypto support + PCB manufacturing procedure

danielebruneo, 2023-07-20:
Hi,
I've been following your project and loving all the work you've done.

This overlap perfectly with a project I was researching on: building a crypto transciever based on an ESP (something similar to what aarmono did here: https://github.com/aarmono/crypto_transceiver_instructions but cheaper and smaller).

I would then like to start contributing to your project, maybe trying to port the wark aarmono did on top of codec2 to build encryption support. (Is this something you were already thinking about, maybe?)

First thing, though, I'd need to get some hardware to develop with.

Any chance there is a ready-to-go package or instructions to get a set of pcb manufactured by any PCBA provider?

oh2lak, 2023-07-20:
What is the need for encryption, as amateur radio traffic in 99% of the
countries do not allow encryption?

Br,
Erik OH2LAK

tmiw, 2023-07-20:
The BOM is built into the KiCad files, but there are specific settings you have to choose at JLCPCB to have it built properly. I'll write up a document on that when I have time (likely not until at least next week).

As for crypto support, I'll leave it to @danielebruneo's fork as ezDV itself is intended for amateur radio use.

tmiw, 2024-05-18:
Closing as dupe of #21.

closed 2024-05-18 by tmiw