tmiw/ezDV #22: WiFi unable to disable

g0kao, 2023-09-05:
Testing the ezDV module using two radios, one IC-705 the other a Yaesu FTdx10, Wifi enabled for the 705 which seems to be working ok, but when trying to focus efforts on the dx10, am unable to disable the WiFi connectivity for the radio.
Whenever i untick the option for WiFi connectivity and try to save, it reverts back to previous settings without unticking the WiFi option.
Have to disable WiFi on the 705 to force connectivity to work with the dx10
Have updated to latest version of code just released via ota but no change in the WiFi issue

tmiw, 2023-09-05:
Do you see any red or green popup boxes on the page after you press Save? If you see a green "success" type message, keep in mind that Wi-Fi changes require a power cycle (or pushing the Reboot button in the web UI) to take effect.

g0kao, 2023-09-05:
I untick the wifi check box, the config clears, I click the Save button, the module resets itself
when it comes back online the wifi is still enabled.
I did an ota update with the new release but it didn.t change anything, should I try a fresh full reflash perhaps?

g0kao, 2023-09-05:
I reflashed the module as discussed, I haven't connected to the ic-705, i just enabled the wifi, reset the module, then when it came back I attempted to disable wifi to the radio and the same steps occurred as outlined above, wifi not disable to the radio. going to try something else like disabling wifi completely from the module as a complete reset of the wifi code and see what happens

g0kao, 2023-09-05:
Nope, cannot disable wifi either, very strange, if i click the reset button on the wifi radio page, it defaults to some preconfigured default info, the black circle continues circling with no other prompts, it appears that once configured it cannot be undone without a reflash for me anyway. I  have one last ditch idea, will let you know what happens

g0kao, 2023-09-05:
attached is a putty log, tried to disable radio wifi, didn't work, then tried to disable wifi as a service, that did not work either
[putty-ezDV-wifi-issue.txt](https://github.com/tmiw/ezDV/files/12525971/putty-ezDV-wifi-issue.txt)

tmiw, 2023-09-06:
Give tonight's release a shot and let me know how it goes: https://github.com/tmiw/ezDV/releases/tag/20230905

g0kao, 2023-09-06:
Thanks Mooneer, that has fixed the problem, I can disable wifi or the wifi radio connection and re-enable the radio connection as needed, thanks again

closed 2023-09-06 by g0kao