tmiw/ezDV #23: revised enclosure top and buttons

g0kao, 2023-09-24:
Having had the enclosure built for me I discovered that when the module v0.7 was fitted, the top did not fit due to the headers fitted to the pcb, as a result I have adjusted the height of the top and the length of the buttons to cater for the new case height.
Attached are new scad files that I have created and successfully printed to fit this module.
enclosure-lid.scad
enclosure-button-v3.scad
[enclosure-new-files.zip](https://github.com/tmiw/ezDV/files/12709445/enclosure-new-files.zip)

tmiw, 2023-09-25:
Thanks for these. How are you routing the cables attached to the headers out? I didn't see any additional holes when I opened them in OpenSCAD.

g0kao, 2023-09-25:
On mine I haven't created any holes for leads to come out from headers, managed connectivity via audio TRRS sockets for DX10 and wifi for ic-705. When original case was made I drilled a hole in the lid directly above the dc connection for the battery and chopped the header pins down so the lid would fit enough with the help of electrical tape to hold it together :-)
Need to make a fix so that the 3d printed buttons don't fall out when the module is tipped over, I think some tiny rubber o-rings should do it if I can find some small enough :-)
Attached is pic of my completed module using the files uploaded on my 3d printer
[image]

tmiw, 2023-09-25:
Ah, okay. I actually intended the headers to be optional since most will either be using the 3.5mm jacks and/or Wi-Fi, but it won't hurt to update the main OpenSCAD files with an extra flag that can be set to print the bigger case.

Re: O-rings, that sounds like a pretty good idea. I was thinking post-processing by adding tiny dabs of glue to the bottoms of the buttons but that precludes removing them later :)

tmiw, 2023-09-26:
I integrated the changes you provided into the main `enclosure.scad` file. The taller enclosure can be requested by setting the below (near the top of the file) to true: 

```
// Whether to print a taller enclosure to fit e.g. pin headers.
// Note: this weirdly has to be above the include line for some reason.
printTallerEnclosure = false;
```

and likewise for `enclosure-button.scad`:

```
// Whether to print a taller buttons (typically used when printing the taller enclosure).
printTallerButtons = false;
```

Let me know if that doesn't work for you and I can reopen and investigate further.

closed 2023-09-26 by tmiw