tmiw/ezDV #24: System stuck if no MAX17048 is used

onemikedelta, 2023-11-01:
I planned to use this module without battery. Everything works however, if I try to connect web interface system reboots. 
Library shouldn't try to read battery if ic is not found during boot process.
`
I (22945) ButtonArray: Button VolDown now 0
I (23925) SettingsTask: Committing pending settings to flash.
I (40405) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (40405) wifi:station: 12:34:56:78:90:AB join, AID=1, bgn, 20
I (40405) WirelessTask: Wifi event: 14
I (40535) wifi:<ba-add>idx:2 (ifx:1, 12:75:8d:8e:1d:67), tid:0, ssn:0, winSize:64
I (40745) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (40765) DVTask: Task FlexVitaTask started
I (57075) HttpServerTask: Sending file : /bootstrap.min.css.gz (27324 bytes)...
I (57195) HttpServerTask: File sending complete
I (57285) HttpServerTask: Sending file : /localstyle.css (3249 bytes)...
I (57295) HttpServerTask: File sending complete
I (57385) HttpServerTask: Sending file : /bootstrap.bundle.min.js.gz (23265 bytes)...
I (57485) HttpServerTask: File sending complete
I (57495) HttpServerTask: Sending file : /jquery-3.6.1.min.js.gz (30850 bytes)...
I (57625) HttpServerTask: File sending complete
I (57715) HttpServerTask: Sending file : /localscript.js (22156 bytes)...
I (57845) HttpServerTask: File sending complete
I (57885) HttpServerTask: Websocket connection opened
I (57885) HttpServerTask: Sending JSON message to socket 57
I (57885) HttpServerTask: Sending JSON message to socket 57
I (57885) HttpServerTask: Sending JSON message to socket 57
I (57895) HttpServerTask: Sending JSON message to socket 57
I (57895) HttpServerTask: Sending JSON message to socket 57
I (57895) HttpServerTask: Sending JSON message to socket 57
I (57905) HttpServerTask: Sending JSON message to socket 57

assert failed: void ezdv::driver::MAX17048::onRequestBatteryStateMessage_(ezdv::task::DVTask*, ezdv::driver::RequestBatteryStateMessage*) MAX17048.cpp:215 (success)

Backtrace: 0x40376016:0x3fca9000 0x40380f31:0x3fca9020 0x40388c26:0x3fca9040 0x42015eca:0x3fca9160 0x42127fca:0x3fca91e0 0x42016639:0x3fca9200 0x42016649:0x3fca9230 0x420162a9:0x3fca9250 0x420162bf:0x3fca9280 0x42022369:0x3fca92a0 0x420223d6:0x3fca92f0 0x42022415:0x3fca9310 0x4038459d:0x3fca9330

------------------
I (1202) MAX17048: Checking for device...
W (1302) MAX17048: Device not found; battery monitoring disabled.
I (1302) DVTask: Task MAX17048 started
I (1312) DVTask: Task LedArray started
I (1322) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1322) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1322) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1322) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1322) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1322) DVTask: Task ButtonArray started
I (1332) TLV320Driver: reset TLV320
I (1332) TLV320Driver: configure clocks
I (1342) TLV320Driver: configure processing blocks
I (1342) TLV320Driver: configure LDOs
I (1342) TLV320Driver: configure routing for ADC
I (1352) TLV320Driver: configure routing for DAC
I (1602) ButtonArray: Button UsbPower now 1
I (3182) TLV320Driver: configuring interrupts
I (3182) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3182) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3182) TLV320Driver: all audio codec config complete
I (3182) DVTask: Task TLV320Driver started
W (3202) TLV320Driver: ADC overload! (Left/User = 1, Right/Radio = 0)
I (3292) DVTask: Task FreeDVTask started
I (3302) DVTask: Task AudioMixer started
I (3302) TLV320Driver: button state: 10
I (3312) DVTask: Task BeeperTask started
I (3322) VoiceKeyerTask: Starting VoiceKeyerTask
I (3322) DVTask: Task VoiceKeyerTask started
I (3332) UserInterfaceTask: Sending welcome message to beeper
I (3332) DVTask: Task UserInterfaceTask started
I (3332) BeeperTask: Got beeper message   ANA
I (3342) DVTask: Task WirelessTask started
I (3342) DVTask: Task MainApp started
I (3352) SettingsTask: Initializing NVS.
I (3352) SoftwareUpdateTask: Starting SoftwareUpdateTask
I (3352) DVTask: Task SoftwareUpdateTask started
I (3352) SettingsTask: Opening NVS handle.
I (3352) SettingsTask: leftChannelVolume: 10
I (3352) SettingsTask: rightChannelVolume: -35
I (3352) SettingsTask: wifiEnabled: 0
I (3352) SettingsTask: wifiMode: 0
I (3352) SettingsTask: wifiSecurity: 0
I (3352) SettingsTask: wifiChannel: 1
I (3352) SettingsTask: wifiSsid:
I (3362) TLV320Driver: Volume control: setting volume on register 65 to 10
I (3362) SettingsTask: wifiPassword: ********
I (3362) SettingsTask: radioEnabled: 0
I (3362) SettingsTask: headsetPtt: 0
I (3362) SettingsTask: timeOutTimer: 120
I (3362) SettingsTask: radioType: 0
I (3362) SettingsTask: radioHostname:
I (3362) SettingsTask: radioPort: 0
I (3362) SettingsTask: radioUsername:
I (3362) SettingsTask: radioPassword: ********
I (3362) UserInterfaceTask: Disallowing headset button from toggling PTT.
I (3362) SettingsTask: enableVoiceKeyer: 0
I (3362) SettingsTask: voiceKeyerNumberTimesToTransmit: 10
I (3362) SettingsTask: voiceKeyerSecondsToWaitAfterTransmit: 5
I (3362) SettingsTask: callsign:
I (3362) SettingsTask: gridSquare: UN00KN
I (3362) SettingsTask: ledDutyCycle: 8192
I (3362) SettingsTask: lastMode: 3
I (3362) DVTask: Task SettingsTask started
I (3362) FreeDV: Setting FreeDV mode to 3
I (3362) BeeperTask: Got beeper message   1600
I (3372) TLV320Driver: Volume control: setting volume on register 66 to -35

`

tmiw, 2023-11-01:
ezDV is intended for use with a battery. Did you simply not populate the chip when you built it, or did you make additional changes?

Regardless, there is logic in there to detect the presence of the MAX17048 but the result of that is apparently getting ignored. That should hopefully be a quick fix that I can get to tonight.

tmiw, 2023-11-02:
This should be fixed in 2f396cb. Please confirm and I can go ahead and close. Thanks for letting me know!

onemikedelta, 2023-11-02:
You are welcome, thank you for such a fast action. It works normal now. I didn't populated the chip on one pcb. 
I tryed to solder on another pcb but it completely stopped booting.

`I (668) sleep: Configure to isolate all GPIO pins in sleep state
I (669) sleep: Enable automatic switching of GPIO sleep configuration
I (669) app_start: Starting scheduler on CPU0
I (669) app_start: Starting scheduler on CPU1
I (669) main_task: Started on CPU0
I (679) esp_psram: Reserving pool of 63K of internal memory for DMA/internal all                                     ocations
I (679) main_task: Calling app_main()
I (779) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (779) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (779) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (779) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (779) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (779) app: Wakeup reason: 0
I (779) app: Power up mode: 2773393973
I (779) app: Woken up via ULP, booting...
I (779) main_task: Returned from app_main()
I (789) app: onTaskStart_
I (809) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (809) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (809) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (809) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (809) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (809) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (809) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo                                     wn: 0| Intr:0
I (809) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (819) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (819) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                     n: 0| Intr:0
I (829) MAX17048: Checking for device...
E (1829) DVTask: Was waiting for MAX17048 but timed out
W (1929) MAX17048: Device not found; battery monitoring disabled.
I (1929) DVTask: Task MAX17048 started

assert failed: void ezdv::task::DVTask::waitForOurs_(ezdv::task::DVTask*, TickTy                                     pe_t) [with ControlMessageType = ezdv::task::TaskControlCommon<3>; TickType_t =                                      long unsigned int] DVTask.h:346 (found)

Backtrace: 0x4037677a:0x3fcbca90 0x4038124d:0x3fcbcab0 0x40388faa:0x3fcbcad0 0x4                                     2021cdd:0x3fcbcbf0 0x42021d59:0x3fcbcc20 0x4200ac8c:0x3fcbcc60 0x42020e96:0x3fcb                                     cca0 0x42122b9d:0x3fcbccd0 0x42020559:0x3fcbcd00 0x4202111a:0x3fcbcd30 0x4202122                                     2:0x3fcbcd80 0x4202127d:0x3fcbcda0 0x40381c01:0x3fcbcdc0
`

tmiw, 2023-11-02:
I added a bit more timeout for the MAX17048 task based on my testing here with an earlier version that doesn't have the MAX17048. Hopefully that helps!

onemikedelta, 2023-11-02:
Now the battery part works perfect. Thank you again, 73

closed 2023-11-02 by onemikedelta