tmiw/ezDV #25: Analog Bypass Feature

onemikedelta, 2023-11-29:
Analog Bypass
If we switch IN1_L and IN1_R in the future versions, it looks like we may have a low power analog bypass mode, which will reduce latency and we can put esp32 into sleep mode to save power.

We may add speaker amplifier to LOL line and internal mic to one of the available inputs, (preferably analog routable one). So, the device can be usable as an ordinary hand mike. I have added LM4871MX on my pcb and routed LOL to this pin, it works nice. We only need to implement some amplifier enable signal other than Sync output because this output flickers a lot and continuously turn amplifier on and off. I have tryed headphone detect interrupt to switch headphone/speaker channels however it is not working, probably allways connected radio channel is preventing that, not sure atm.

Just a few ideas, 73

tmiw, 2023-11-29:
I remember the TLV320 being pretty low power even without using its analog bypass mode. The ESP32-S3 (and the status LEDs to a smaller extent) by far use most of the power. That said, using the mixer amps may be possible for RX; see below from the Application Reference Guide:

[image]

(Basically, set HPL to MAR, which comes from IN1_R.)

However, on TX, the ESP32 would probably have to come back online to take back control of audio routing, which will take a second or two if coming from ULP sleep. Not sure users would be okay with that, to be honest.

onemikedelta, 2023-11-29:
The problem is we cannot route microphone to the radio in this configuration if I am not missing a path. MAL to HPR is not supported. For the PTT, we may introduce a simple diode from radio PTT to PTT button. Only thing that is not going to work is PTT LED. I haven't tryed to setup interrupt for button and use GPIO21 yet. Maybe we can achieve this ptt system without waking up mcu completely by using ULP.

tmiw, 2023-11-29:
I don't have the schematic in front of me but I think the PTT output pin is one that can be manipulated in ULP. Would just need hardware changes to enable the proper routing if you did want to be able to support low powered analog.

That said, switching from that mode would likely still take a few seconds, but that might still be okay. Let's put this on hold for now and consider for ezDV V2.