tmiw/ezDV #59: ezDV firmware build issues

FreeRepositories, 2024-07-29:
Hi,
Trying to build the firmware,  (commit 7ee14a9 v1.1.0)  but it looks it cannot be built.
Seen that in Releases there is a pre-built firmware, I wonder what is happening.
Are the repo sources checked out, up to date ?

Tried this many times, with the same error results.

New environment, Ubuntu 24.04, fresh esp-idf v 5.3

In a nutshell, what is executed is

**git clone https://github.com/tmiw/ezDV.git**
**cd ezDV**
**git submodule update --init --checkout --recursive**
**cd firmware**
**idf.py build**

 and then it stops with 

```
/home/user/esp/projects/ezDV/firmware/main/network/HttpServerTask.cpp:612:28: error: 'using ezdv::network::HttpServerTask::WebSocketList = class std::map<int, bool>' {aka 'class std::map<int, bool>'} has no member named 'contains'
  612 |     if (!activeWebSockets_.contains(message->fd))
      |                            ^~~~~~~~
/home/user/esp/projects/ezDV/firmware/main/network/HttpServerTask.cpp: In member function 'void ezdv::network::HttpServerTask::onHttpWebsocketDisconnectedMessage_(ezdv::task::DVTask*, HttpWebsocketDisconnectedMessage*)':
/home/user/esp/projects/ezDV/firmware/main/network/HttpServerTask.cpp:840:27: error: 'using ezdv::network::HttpServerTask::WebSocketList = class std::map<int, bool>' {aka 'class std::map<int, bool>'} has no member named 'contains'
  840 |     if (activeWebSockets_.contains(message->fd))
      |                           ^~~~~~~~
```

Here is the complete output from the clone and build process

```
user@user-VMware-Virtual-Platform:~/esp/projects$ git clone https://github.com/tmiw/ezDV.git
Cloning into 'ezDV'...
remote: Enumerating objects: 7373, done.
remote: Counting objects: 100% (625/625), done.
remote: Compressing objects: 100% (279/279), done.
remote: Total 7373 (delta 473), reused 468 (delta 346), pack-reused 6748
Receiving objects: 100% (7373/7373), 104.54 MiB | 6.18 MiB/s, done.
Resolving deltas: 100% (5528/5528), done.
user@user-VMware-Virtual-Platform:~/esp/projects$ cd ezDV
user@user-VMware-Virtual-Platform:~/esp/projects/ezDV$ git submodule update --init --checkout --recursive
Submodule 'enclosure/library/YAPP_Box' (https://github.com/mrWheel/YAPP_Box.git) registered for path 'enclosure/library/YAPP_Box'
Submodule 'firmware/externals/esp-dsp' (https://github.com/espressif/esp-dsp.git) registered for path 'firmware/externals/esp-dsp'
Submodule 'firmware/externals/tinyuntar' (https://github.com/tmiw/TinyUntar.git) registered for path 'firmware/externals/tinyuntar'
Submodule 'firmware/externals/uzlib' (https://github.com/tmiw/uzlib) registered for path 'firmware/externals/uzlib'
Cloning into '/home/user/esp/projects/ezDV/enclosure/library/YAPP_Box'...
Cloning into '/home/user/esp/projects/ezDV/firmware/externals/esp-dsp'...
Cloning into '/home/user/esp/projects/ezDV/firmware/externals/tinyuntar'...
Cloning into '/home/user/esp/projects/ezDV/firmware/externals/uzlib'...
Submodule path 'enclosure/library/YAPP_Box': checked out 'b18e996388dff5b3ce523cec2b26665ccb7059d4'
Submodule path 'firmware/externals/esp-dsp': checked out '71514173b58b960173b40c4ade9d15d372770a74'
Submodule path 'firmware/externals/tinyuntar': checked out 'bda54b1c12141c74f259b4893186556a72c0d052'
Submodule path 'firmware/externals/uzlib': checked out '6ffe173438baa8a41bb96e5ab658e9c254a4bcac'
user@user-VMware-Virtual-Platform:~/esp/projects/ezDV$ cd firmware
user@user-VMware-Virtual-Platform:~/esp/projects/ezDV/firmware$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/user/esp/projects/ezDV/firmware/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/user/.espressif/python_env/idf5.3_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/user/esp/projects/ezDV/firmware"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/home/user/esp/projects/ezDV/firmware/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
[... 409 lines cut ...]
      | ^
[1405/1426] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/network/HttpServerTask.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/network/HttpServerTask.cpp.obj 
/home/user/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -DCODEC2_MODE_1300_EN=1 -DCODEC2_MODE_700C_EN=1 -DCODEC2_MODE_EN_DEFAULT=0 -DESP_PLATFORM -DFREEDV_MODE_1600_EN=1 -DFREEDV_MODE_700D_EN=1 -DFREEDV_MODE_700E_EN=1 -DFREEDV_MODE_EN_DEFAULT=0 -DIDF_VER=\"v5.3\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -D__EMBEDDED__ -D__REAL__ -I/home/user/esp/projects/ezDV/firmware/build/config -I/home/user/esp/projects/ezDV/firmware/main -I/home/user/esp/esp-idf/components -I/home/user/esp/projects/ezDV/firmware/main/../externals/tinyuntar/tinyuntar -I/home/user/esp/projects/ezDV/firmware/main/../externals/uzlib/src -I/home/user/esp/projects/ezDV/firmware/build/esp-idf/main/ulp_main -I/home/user/esp/esp-idf/components/newlib/platform_include -I/home/user/esp/esp-idf/components/freertos/config/include -I/home/user/esp/esp-idf/components/freertos/config/include/freertos -I/home/user/esp/esp-idf/components/freertos/config/xtensa/include -I/home/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/user/esp/esp-idf/components/freertos/esp_additions/include -I/home/user/esp/esp-idf/components/esp_hw_support/include -I/home/user/esp/esp-idf/components/esp_hw_support/include/soc -I/home/user/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/home/user/esp/esp-idf/components/esp_hw_support/dma/include -I/home/user/esp/esp-idf/components/esp_hw_support/ldo/include -I/home/user/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/home/user/esp/esp-idf/components/esp_hw_support/port/esp32s3/include -I/home/user/esp/esp-idf/components/heap/include -I/home/user/esp/esp-idf/components/log/include -I/home/user/esp/esp-idf/components/soc/include -I/home/user/esp/esp-idf/components/soc/esp32s3 -I/home/user/esp/esp-idf/components/soc/esp32s3/include -I/home/user/esp/esp-idf/components/hal/platform_port/include -I/home/user/esp/esp-idf/components/hal/esp32s3/include -I/home/user/esp/esp-idf/components/hal/include -I/home/user/esp/esp-idf/components/esp_rom/include -I/home/user/esp/esp-idf/components/esp_rom/include/esp32s3 -I/home/user/esp/esp-idf/components/esp_rom/esp32s3 -I/home/user/esp/esp-idf/components/esp_common/include -I/home/user/esp/esp-idf/components/esp_system/include -I/home/user/esp/esp-idf/components/esp_system/port/soc -I/home/user/esp/esp-idf/components/esp_system/port/include/private -I/home/user/esp/esp-idf/components/xtensa/esp32s3/include -I/home/user/esp/esp-idf/components/xtensa/include -I/home/user/esp/esp-idf/components/xtensa/deprecated_include -I/home/user/esp/esp-idf/components/lwip/include -I/home/user/esp/esp-idf/components/lwip/include/apps -I/home/user/esp/esp-idf/components/lwip/include/apps/sntp -I/home/user/esp/esp-idf/components/lwip/lwip/src/include -I/home/user/esp/esp-idf/components/lwip/port/include -I/home/user/esp/esp-idf/components/lwip/port/freertos/include -I/home/user/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/user/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/user/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/user/esp/projects/ezDV/firmware/externals/esp-dsp/modules/dotprod/include -I/home/user/esp/projects/ezDV/firmware/externals/esp-dsp/modules/support/include -I/home/user/esp/projects/ezDV/firmware/externals/esp-dsp/modules/support/mem/include -I/home/user/esp/projects/ezDV/firmware/externals/esp-dsp/modules/windows/include -I/home/user/esp [... cut ...]
      |                                                                                                     ^~
[1407/1426] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/network/PskReporterTask.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/user/esp/projects/ezDV/firmware/build/log/idf_py_stderr_output_12757 and /home/user/esp/projects/ezDV/firmware/build/log/idf_py_stdout_output_12757
user@user-VMware-Virtual-Platform:~/esp/projects/ezDV/firmware$ 
```

tmiw, 2024-07-29:
The prebuilt firmware images and an initial release record are generated via GH actions upon tagging a new release (and whether PRs build is also one of the things checked for before merging). The build output from the 1.1.0 release build can be found at https://github.com/tmiw/ezDV/actions/runs/10148059074/job/28059905437.

As for your issue, how did you install ESP-IDF? It might be worth it to reinstall it following the instructions at https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32s3/get-started/index.html (possibly in a new terminal window). For example:

```
rm -rf ~/esp
mkdir -p ~/esp
cd ~/esp
git clone -b v5.3 --recursive https://github.com/espressif/esp-idf.git
./install.sh all
cd ~/ezDV/firmware
. ~/esp/esp-idf/export.sh
idf.py fullclean build 
```

FreeRepositories, 2024-07-30:
Ok, you're right.
Did another fresh esp-idf install and now it works fine!
After that though, got acquainted with the Actions, never used them in the past..

closed 2024-07-30 by FreeRepositories